Quicr/hactar #40: Make sure audio codec chip can not mess up other chips

fluffy, 2024-02-28:
Right now the serial data out is messing up the auto boot load mode of other things.

fluffy, 2024-02-28:
One possibility is to have and gate on output of codec with a enable line default low and controlled by UI cpu

fluffy, 2024-02-28:
@BrettRegnier - Can you provide more detail about what is going wrong right now

BrettRegnier, 2024-03-22:
The pins that the wm8960 ADC is connect to can also be configured as USART3_RX, which when in bootloader mode all of the possible USART pins are configured into USART mode, and select the first configured USART bus as the communication line. Since the audio chip is always sending data unless it is reset then the bootloader will attempt to communicate using USART3 instead of USART1.

So, all we need to do is change which pin our i2s is connect to by swapping PC11 to PB4 (I2S_ADCDAT)

fluffy, 2024-03-25:
Did this happen on 10.1 ?

BrettRegnier, 2024-04-22:
Yes! This is fixed in 10.1

closed 2024-04-22 by BrettRegnier