Quicr/hactar #121: Pull up and down for MGMG NRST/Boot

fluffy, 2025-01-14:


BrettRegnier, 2025-02-25:
I am not sure if we need to add an external pull-up resistor to the nrst for mgmt. We are currently using their recommended external reset circuit.

pg 84 of the stm32f702 [datasheet](file:///home/brett/Downloads/stm32f072c8-1.pdf)
[image]

BrettRegnier, 2025-02-25:
As well, the mgmt_boot has a pulldown resistor.

[image]

BrettRegnier, 2025-04-09:
@fluffy
Added pull-down to UI_BOOT0 for normal operation on this [commit](https://github.com/Quicr/hactar/commit/fca9c4b6ff0e0127da8520f6f3ede625ea799450)

fluffy, 2025-04-17:
LGTM