Quicr/hactar #144: Add mount holes to PCB

fluffy, 2025-01-18:
Prob set up for M2.5 bolts

BrettRegnier, 2025-02-24:
Do we have specific spots we want them?

fluffy, 2025-04-09:
Well roghly speaking withing 2 cm of each corrner. But they need to be more than 3 mm from the edge and we have lots of other stuff we need to avoid

BrettRegnier, 2025-05-29:
The mount hole near the boost converter is 1mm too close to the edge

[image]

The mount hole near the headphones jacks is about 0.5mm too close to the edge.

[image]

Mount hole near the top right is about 0.5mm too close

[image]

Same with the top left.

[image]