Quicr/hactar #156: Add schematic note on MGMT boot pins

fluffy, 2025-02-16:
Add to the schematic pins that have to be stable at boot to not be detected as bootloader input

PA9/PA10 is used for actually bootload

PA15 - input usart2 
PA14 - output usart2

PB6,PB7 - in/out for I2C 

PA11/PA12 - in/out for USB 

Notes on current setup 

PA14,PA15 prob need pull down. 

PB6/PB7 - need pull down restores - currently floating 

PA11,12 - need pull down resistors - currently floating.

fluffy, 2025-02-16:
From datasheet

> The boot loader is located in System Memory. It is used to reprogram the Flash memory by
> using USART on pins PA14/PA15, or PA9/PA10 or I2C on pins PB6/PB7 or through the USB
> DFU interface.

Finnberg9, 2025-05-15:
Note that on EV13, the pins are totally different. TO solve this issue, I added 4.7k pull down resistors to all unused GPIO pins on the STM32F0. These pins were PA5, PB12, and PC13

BrettRegnier, 2025-05-29:
I think it would be helpful to put "bootloader" on each pin like we have on the ui so its more clear and easier to hunt down the pins than reading the note.

BrettRegnier, 2025-05-29:
Otherwise, I think it looks fine

closed 2025-07-25 by BrettRegnier