Quicr/hactar #168: Feed barCHG from charger to MGMT chip

fluffy, 2025-02-16:
Allow MGMT chip to monitor if charged or not

Finnberg9, 2025-05-13:
Add trace from MGMT GPIO to CHG line between diode and resistor

fluffy, 2025-07-26:
Could PC13 or PB1 be used ?

fluffy, 2025-11-24:
The more I think about it, the more I think it would be better to just feed this to the UI chip rather than mgmt. In the original design we had the mgmt chip could control the power mode of things and push the UI / net into low power mode and manage taking them out but I think we can do all of that from software on UI chip so rather just keep the mgmt chip to have most minimal of functionality

bifurcation, 2025-11-24:
UI chip logic is a precious commodity, and there's no reason to have power logic on the UI chip.  Just have the NET chip do the measurements and analysis and tell the UI chip what to display.

BrettRegnier, 2025-11-25:
Here is my circuit using a mosfet as a switch

[image]

closed 2025-12-17 by BrettRegnier