Quicr/hactar #170: Stable boot pins during bootload

fluffy, 2025-02-16:
PA9/10 used for bootloader

PB10/11 USART3  - 10 in , 11 out
PC10/11 USART3 - 10 in 11 out 
PB5/13 CAN 5 in , 13 out 
PA11/12 USB  - in and out

fluffy, 2025-02-16:
PB 11 likely floating to keyboard 
PC 10,11 are floating 
PB 5,13 are floating 
PA 11,12 are floating

fluffy, 2025-02-16:
From datasheet

> The boot loader is located in system memory. It is used to reprogram the flash memory by
> using USART1 (PA9/PA10), USART3 (PC10/PC11 or PB10/PB11), CAN2 (PB5/PB13), USB
> OTG FS in Device mode (PA11/PA12) through DFU (device firmware upgrade).

BrettRegnier, 2025-02-19:
Updated in [commit](https://github.com/Quicr/hactar/commit/27ad6a8fa451261320cadf4bf07ff31d7a957172)