Quicr/hactar #182: Is kbd connector too close to battery ?

fluffy, 2025-02-20:


BrettRegnier, 2025-02-20:
[image]

Looks like it fits pretty good! Tight, but the battery and connector don't touch

fluffy, 2025-02-23:
Uh, I don't think flex cables are supposed to have that tight of a bend anywhere on them. It should come more straight of the connector then slowly flex. We can try to find the specs on flex but doubt that is ok

BrettRegnier, 2025-02-25:
In the photo I did exaggerate the bend in the flex cable to get a good shot. Regardless, it does have bend we may need to reconsider it's position in relation to where the keyboard will be sitting with a case. Otherwise, the bend isn't too sharp.

Finnberg9, 2025-05-13:
Move keyboard connector down as much as possible

closed 2025-05-14 by Finnberg9