Quicr/hactar #190: Move MIC_P to LINPUT3 so we can use jack detect

BrettRegnier, 2025-02-27:
Assuming @fluffy agrees.

fluffy, 2025-02-28:
I want to review this really carefully before we do it. I'm not sure it will do what you want to reduce clicks and I'm actually more keen on moving to the mics with a PTT button. It's just I think we put a bunch of thought into this before

BrettRegnier, 2025-02-28:
As I read more about it, the big benefit I would see for it is that we could keep our audio levels at zero, and when the headphone jack detect signal is sent we can raise our output volumes to the headphones to try and avoid the "DC pop".

Alternatively, according to this forum we could try adding a RC circuit on the output to a speaker, I don't know if this would also work for a headphone or not.

https://electronics.stackexchange.com/questions/323356/loud-pop-noise-while-plugging-to-a-headphones-jack

BrettRegnier, 2025-04-10:
From the WM8960 chip, before or after the capacitor?

fluffy, 2025-04-14:
after the capacitor

fluffy, 2025-05-01:
LGTM

closed 2025-05-01 by fluffy