Quicr/hactar #196: Move LEDs to make sense from UI point of view

fluffy, 2025-04-09:
Right now it is hard to even describe to people which LED is which and the pairs are so close togehter it is sometimes hard to know which is which. I'm not sure exactly what we should do here but seems like we need a bit of design of where LEDs are on case

vcostama, 2025-04-23:
[image]

this is current state of the led lights, we'll be updating the position of the led lights when Brett returns from pto. i'll also be updating this docu audio to compliment the leds.

fluffy, 2025-04-25:
Sounds good. 

Eventually, I suspect we need indicate at least the following:

Software Error
No Wifi Connection
No Relay Connection 
Low Battery 

Later we might want to indicate:

Message waiting 

I'm not sure what else is needed but would want over time.

vcostama, 2025-04-25:
thanks for bringing those up. definitely i'll incorporate those items and any additional items we need to convey status/state.

vcostama, 2025-05-15:
[image]

version 2