Quicr/hactar #203: Find new capacitor for power supply

BrettRegnier, 2025-04-23:
Previous capacitor
https://jlcpcb.com/partdetail/C495959

In case the page gets taken down here are the specs.
[image]

BrettRegnier, 2025-04-23:
https://jlcpcb.com/partdetail/28983-CL21B104KCFNNNE/C28233

This seems like it has similar attributes with the same package.

fluffy, 2025-05-01:
Interesting. It looks like EV10 schmatic had a mistake and did not specity a part numnber for that then on a later one the wrong one got added. The cap we had was not a good choice. 

I think we should go with 
https://jlcpcb.com/partdetail/MurataElectronics-GCM21BR72A104KA37L/C85866

BrettRegnier, 2025-05-01:
Ok!

BrettRegnier, 2025-05-05:
#209

closed 2025-05-05 by BrettRegnier