Quicr/hactar #204: Find new ptt buttons

BrettRegnier, 2025-04-23:
Previous buttons
https://jlcpcb.com/partdetail/Yuandi-Yuandi_TS_1037_A11B3D2/C499331

[image]

BrettRegnier, 2025-04-23:
https://jlcpcb.com/partdetail/ShouHan-TS665TPZJ/C557600

If we did the above part I could design some 3d printed buttons that get pinched between the case top and bottom and will wrap around the tip of the button so it doesn't slide around and then it will still feel nice and tactile but we can then design how we want the button look like!

BrettRegnier, 2025-04-30:
One more alternative:
https://jlcpcb.com/partdetail/Yuandi-TS_1045_A13B3D2/C557898

fluffy, 2025-05-01:
I think I lean towards trying the https://jlcpcb.com/partdetail/Yuandi-TS_1045_A13B3D2/C557898 

Do we have stl models for it ?

BrettRegnier, 2025-05-01:
EasyEDA has models so I could download the model

closed 2025-05-05 by BrettRegnier