Quicr/hactar #270: Store config in EEProm

fluffy, 2025-06-14:
- [ ] Make the USB connection so it can run terminal program and store config stuff in EEProm 
- [ ] Store the WiFi SSID and pwd in the EEProm 
- [ ] Store the Sframe Key in EEProm 
- [ ] Store URL of MoQ relay in EEProm

fluffy, 2025-06-14:
If it helps, I have code in other project so to be able to read / write to eeprom we are using from stm32

BrettRegnier, 2025-06-15:
Will we have enough space on our eeprom? It's only 1024 bytes of storage

fluffy, 2025-08-05:
@bifurcation - currently when we upgrade the firmware on the net chip, we erase all the flash and surely you are not suggesting the crypto info is on the flash of the net chip ???

bifurcation, 2025-08-05:
1. We need persistence on the net chip.  If we don't have that today, that's a different problem.
2. We don't have a requirement to program any crypto info right now.